aolofsson/oh #14: missing common/dv folder in commit fb45666b13dcd1ee8e5f88adfe1c0a4cf24d6745

peteasa, 2015-11-03:
Just a quick reminder to push common/dv/dv_top.v and common/dv/libs.c these files do not appear but are required by elink/dv/build.sh

aolofsson, 2015-11-03:
Thanks! Sorry about that! Just added them (I pass partial blame to all the xilinx crud in my tree)

peteasa, 2015-11-03:
Thanks for the updated files... When I run any of the simulations they seem to take a long time ie start goes active @ 1.6us, first access dv_top-> dut finishes @1.7us, second @100us, third @63us etc.. previously with the old tests I got multiple packet transfers in .5us .. not sure why the new simulation runs at this slow rate and produces large simulation output files...  Anyway the simulation works so thanks.

closed 2015-11-03 by peteasa