aolofsson/oh #17: Reading to register bug

olajep, 2015-11-11:
IIRC

[0x8e000000] == 0x87654321
```asm
mov r0, 0
movt r0, 0x8e00
ldrh r0,[r0]
```
This results in r0 now holding 0x8e004321, when it should be 0x00004321.

olajep, 2015-11-11:
Same applies for bytes

aolofsson, 2015-11-11:
I don't understand the test case/expected result? Please describe the
sequence of events.

olajep, 2015-11-11:
Last line should read
This results in r0 now holding 0x87654321, when it should be 0x00004321.

1. load turns into read access on mesh network
2. packet reaches fpga elink
3. fpga elink sends back write packet w/ higher bits not zeroed out ?
4. packet reaches requesting core and wrong value gets written to register

olajep, 2015-11-11:
To clarify: This is unrelated to the remapping bug.

olajep, 2015-11-30:
fixed

closed 2015-11-30 by olajep