aolofsson/oh #19: writes to fpga regs during reset can get lost

olajep, 2015-11-11:
1. assert reset bit
2. deassert reset bit
3. write something to RXCFG register
= write gets lost ( RXCFG = 0)

1. assert reset bit
2. deassert reset bit
3. usleep(1000)
4. write something to RXCFG register
= write gets through

Isn't there a AXI wait signal (WREADY?) we can use?

Alternatively can we do a read loop in sw over the reset register until the reset bit reads back as de-asserted?

or make it mandatory to sleep after the reset bit has been deasserted

aolofsson, 2015-11-11:
There is a tx_active signal. Let's see if we can tie this back to make write wait....
In the meantime, usleep(1000)..

olajep, 2015-11-11:
Posting again as github screwed up the formatting in the first try.

#### Case 1
1. assert reset bit
2. deassert reset bit
3. write something to RXCFG register

= write gets lost ( RXCFG = 0)

#### Case 2
1. assert reset bit
2. deassert reset bit
3. usleep(1000)
4. write something to RXCFG register

= write gets through

Isn't there a AXI wait signal (WREADY?) we can use?

Alternatively can we do a read loop in sw over the reset register until the reset bit reads back as de-asserted?

or make it mandatory to sleep after the reset bit has been deasserted