aolofsson/oh #20: mailbox read timing issue

peteasa, 2015-11-15:
I created an axi mailbox dut and tried out the same read test I tried earlier and still found the same problems
[image]

Description:  erx_cfgif raises mi_match correctly and mi_cfg_en is raised so that the mailbox read cycle is started, however the mailbox fifo only outputs the data on the rising edge of erx_mailbox.rd_clk and at the same time the erx_mailbox.mailbox_fifo_data with the old content gets sampled and sent to mi_dout.  This can be seen in the attached diagram because the mailbox_fifo_data to the left of the marker gets sampled in the rising edge of the clock and output on mi_dout, however on the next clock cycle the correct data is available on mailbox_fifo_data.

Potential fix is to create a erx_mailbox valid signal emailbox_access_out that must then delay erx_cfgif.access_out so that the correct mailbox data gets delivered to the rxrr fifo.  Last time I tried this fix the simulations failed because changing ecfg_if to take account of the emailbox_access_out affects the etx_cfgif block also. So it is a large change.  Perhaps now it is time to introduce this valid signal for all the ecfg_if.mi_doutX channels?

peteasa, 2015-11-15:
provided pull request that creates what might turn out to be a quick fix.  worried about the delays introduced by the various mux components in the timing chain.. may not work after synthesis

peteasa, 2015-11-28:
Now that the fpga can be loaded into the parallella board I have been trying to continue to get the mailbox working with little success so far.  I will go back to simulation in a while.. but perhaps you can tell me if I am wrong with the following:

I have an e_mailbox.c file that gets built and loaded onto the Epiphany cores.  Just like the Hello_World example.  In the Epiphany I think that I should be able to read the axi_elink registers using:

```C
#define ELINK_BASE	0x81000000
#define E_MAILBOXLO	0xF0310
...
	// Write coreid to the mailbox
	e_coreid_t * dst;
	dst = (e_coreid_t *)(ELINK_BASE | E_MAILBOXLO);
	*dst = coreid;
```
From Epiphany I should be able to read the ERX_CFG register and others as well.

This does not seem to work just yet.

Oh the host side I should be able to read the ELINK_MAILBOXHI, then the ELINK_MAILBOXLO should contain the value I stored in the Epiphany code
```C
	unsigned int mailboxEntry;
	e_debug_read(ELINK_MAILBOXHI, &mailboxEntry);
	printf("HI: 0x%x: 0x%x;  ", ELINK_MAILBOXHI, mailboxEntry);
	e_debug_read(ELINK_MAILBOXLO, &mailboxEntry);
	printf("LO: 0x%x: 0x%x\n", ELINK_MAILBOXLO, mailboxEntry);
```

I can read the registers on the host side and for example I can read the TXCFG register and get 0x1 but when I read this from the Epiphany side I just get 0x0...

Is there something wrong with my software?

aolofsson, 2015-11-29:
Sorry, you can't read the ELINK registers from the Epiphany. They are only accessible from the host. Only the mailbox is writeable (but not readable) from the  Epiphany.

It would have complicated the design and I felt that it wasn't a must have feature.

aolofsson, 2015-11-29:
Fixed the zero order bug on the readback. Once everything else is working, this is a quick fix. (fix is very similiar to 2nd pull request you made). I did not take the pull request as is for style reasons. Hope that's OK, it's just a few lines of code.

Also, there is a silly conceptual bug that I missed. (realized it when I started testing). There is no way to reach 810f0310 from the Epiphany (since this would be routing to the west). I will fix this tomorrow.

peteasa, 2015-11-29:
Not very interested in mailbox readback from epiphany core.  

Sounds like my software should have worked so I will try it out with the latest mailbox hdl later this week.  I put the example software in https://github.com/peteasa/examples/tree/parallella-oh/epiphany/mailbox only slight change from the hello world code for now.  But I am guessing that e-lib will change at some point to add specific api for access to the mailbox, plus some small changes to the epiphany driver code to add interrupt handling.

It was fun to see the hello_world test working with the new oh fpga.

peteasa, 2015-11-29:
Puzzled about the comment about routing to the west...  If the Node address is 0x808 ie col 8 row 32 and the destination is 0x810 ie col 16 row 32 then I would have expected routing east to get closer to the higher col number....  back to the typo in the architecture spec.. I believe the diagrams not the words or the table.

aolofsson, 2015-11-29:
Sorry about confusion. Was tired last night...the problem was in the remap
circuit. About to check in working mailbox..reads have been verified.

peteasa, 2015-11-30:
It works a treat for me also!  Well done!  Each core is reporting its ID via the mailbox with my test app...
# /usr/epiphany/bin/mailbox
main: rows,cols: ( 4, 4)
main:   0: Message from eCore 0x808 ( 0, 0): main: Error in e_load 37056
"Hello World from core 0x808!"
PRE_STAT=00000001 POST_STAT=00000000 LO=00000808 HI=810f0320
PRE_STAT=00000000 POST_STAT=00000000 LO=00000808 HI=810f0320
main:   1: Message from eCore 0x809 ( 0, 1): main: Error in e_load 37056
"Hello World from core 0x809!"
PRE_STAT=00000001 POST_STAT=00000000 LO=00000809 HI=810f0320
PRE_STAT=00000000 POST_STAT=00000000 LO=00000809 HI=810f0320
main:   2: Message from eCore 0x80a ( 0, 2): main: Error in e_load 37056
"Hello World from core 0x80a!"
PRE_STAT=00000001 POST_STAT=00000000 LO=0000080a HI=810f0320
PRE_STAT=00000000 POST_STAT=00000000 LO=0000080a HI=810f0320

closed 2015-11-30 by peteasa