aolofsson/oh #24: latest fails to build

peteasa, 2015-12-03:
ERROR: [Synth 8-439] module 'dsync' not found - sources_1/ipshared/www.parallella.org/parallella_base_v1_0/258848fc/src/emaxi.v:424]
ERROR: [Synth 8-285] failed synthesizing module 'emaxi'
ERROR: [Synth 8-285] failed synthesizing module 'axi_elink'
ERROR: [Synth 8-285] failed synthesizing module 'parallella_base'
ERROR: [Synth 8-285] failed synthesizing module 'system_parallella_base_0_0'
ERROR: [Synth 8-285] failed synthesizing module 'system'
ERROR: [Synth 8-285] failed synthesizing module 'system_wrapper'

aolofsson, 2015-12-04:
I am not seeing that? (even did a clean co to verify).  However, I did just check in a couple of small fixes that popped up. (I am using the "common" folder for some other stuff, and managed to break something). Changes settling down this week...

peteasa, 2015-12-04:
synth working again with latest catchup.

./build.sh works with latest catchup

./build_axi_elink.sh fails:

./build_axi_elink.sh 
../../axi/hdl/esaxi.v:21: warning: macro EGROUP_RR undefined (and assumed null) at this point.
../../axi/hdl/esaxi.v:21: syntax error
../../axi/hdl/esaxi.v:21: error: syntax error in parameter list.
../../axi/hdl/esaxi.v:449: error: Unable to bind wire/reg/memory `RETURN_ADDR' in `dv_top.dut.elink1.esaxi'
../../axi/hdl/esaxi.v:449: error: Unable to bind wire/reg/memory `RETURN_ADDR' in `dv_top.dut.esaxi'
2 error(s) during elaboration.

different issue so I will close this

closed 2015-12-04 by peteasa