aolofsson/oh #25: interrupt handling

peteasa, 2015-12-06:
I now have an updated epiphany driver running in my kernel that handles the interrupts from the mailbox.  I will publish this soon but right now I have a strange problem when using it:
I visit each core in the Epiphany chip and from that core write to the mailbox the core id of the core.
In the driver I map the ERX_CFG registers with devm_ioremap_resource() and in the interrupt handler I simply disable the interrupt in ERX_CFG.  In the near future I will have a sysfs file with the number of messages pending, but for now I keep it simple and from the command line monitor how many interrupts I have seen using the contents of /proc/interrupts.
Before visiting each core I enable the interrupt in the host application using a new ioctl call to the driver.  After the core has sent the message back via the mailbox I read the number of messages with a call to ee_read_esys() and read the message to clear the source of the interrupt.
After the first core it seems that something goes wrong with the loading of the epiphany code.  So I stopped reading the messages till the end of the run and this is what I get:
```C
			int mbox_lo     = ee_read_esys(ELINK_MAILBOXLO);
			int mbox_hi     = ee_read_esys(ELINK_MAILBOXHI);
			int post_stat   = ee_read_esys(ELINK_MAILBOXSTAT);
			printf ("main(): PRE_STAT=%08x POST_STAT=%08x LO=%08x HI=%08x\n", pre_stat, post_stat, mbox_lo, mbox_hi);
```
# /usr/epiphany/bin/mailbox
main: rows,cols: ( 4, 4)
main:   0: Message from eCore 0x808 ( 0, 0): main: Error in e_load 35476
"Hello World from core 0x808!"
main:   1: Message from eCore 0x809 ( 0, 1): main: Error in e_load 35476
"Hello World from core 0x809!"
main:   2: Message from eCore 0x80a ( 0, 2): main: Error in e_load 35476
"Hello World from core 0x80a!"
main:   3: Message from eCore 0x80b ( 0, 3): main: Error in e_load 35476
"Hello World from core 0x80b!"
main:   4: Message from eCore 0x848 ( 1, 0): main: Error in e_load 35476
"Hello World from core 0x848!"
main:   5: Message from eCore 0x849 ( 1, 1): main: Error in e_load 35476
"Hello World from core 0x849!"
main:   6: Message from eCore 0x84a ( 1, 2): main: Error in e_load 35476
"Hello World from core 0x84a!"
main:   7: Message from eCore 0x84b ( 1, 3): main: Error in e_load 35476
"Hello World from core 0x84b!"
main:   8: Message from eCore 0x888 ( 2, 0): main: Error in e_load 35476
"Hello World from core 0x888!"
main:   9: Message from eCore 0x889 ( 2, 1): main: Error in e_load 35476
"Hello World from core 0x889!"
main:  10: Message from eCore 0x88a ( 2, 2): main: Error in e_load 35476
"Hello World from core 0x88a!"
main:  11: Message from eCore 0x88b ( 2, 3): main: Error in e_load 35476
"Hello World from core 0x88b!"
main:  12: Message from eCore 0x8c8 ( 3, 0): main: Error in e_load 35476
"Hello World from core 0x8c8!"
main:  13: Message from eCore 0x8c9 ( 3, 1): main: Error in e_load 35476
"Hello World from core 0x8c9!"
main:  14: Message from eCore 0x8ca ( 3, 2): main: Error in e_load 35476
"Hello World from core 0x8ca!"
main:  15: Message from eCore 0x8cb ( 3, 3): main: Error in e_load 35476
"Hello World from core 0x8cb!"
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=00000808 HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=00000809 HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=0000080a HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=0000080b HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=00000848 HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=00000849 HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=0000084a HI=810f0320
main(): PRE_STAT=00000001 POST_STAT=00000001 LO=0000084b HI=810f0320

peteasa, 2015-12-06:
By the way the number of messages in the mailbox queue might be of use.

aolofsson, 2015-12-06:
From the printout it seems that all the cores are printing back Hello
World. Is the concern "Error in e_load 35476"?
Can you send a pointer to the example code?

peteasa, 2015-12-07:
All code is avialable at https://github.com/peteasa/examples/tree/parallella-oh/epiphany/mailbox
New driver see - https://github.com/peteasa/examples/blob/parallella-oh/epiphany/mailbox/kernel/epiphany.c

I guess this driver already answers the earlier issue you have about crashing when the interrupts are switched on!

New test code see - https://github.com/peteasa/examples/blob/parallella-oh/epiphany/mailbox/host/mailbox.c

I am not too worried about the Error in e_load.. more about the failure of the sequence:
```
ee_write_esys(ELINK_RXCFG, rxcfg | (0x1 << 28))
e_load("/usr/epiphany/bin/e_mailbox.srec", &dev, 0, 0, E_TRUE);
** interrupt occurs and in the handler - cfg = reg_read(&mailbox, ERX_CFG); 	reg_write(&mailbox, ERX_CFG, cfg & ~MAILBOX_ENABLE);
int items = ee_read_esys(ELINK_MAILBOXSTAT);
ee_read_esys(ELINK_MAILBOXLO);
ee_write_esys(ELINK_RXCFG, rxcfg | (0x1 << 28))
e_load("/usr/epiphany/bin/e_mailbox.srec", &dev, 0, 0, E_TRUE);
***** Hmm.. the core load seems to fail at this point ******
***** so no interrupt and no data ****
xx int items = ee_read_esys(ELINK_MAILBOXSTAT);
xx ee_read_esys(ELINK_MAILBOXLO);
```
The following sequence works just fine:
```
ee_write_esys(ELINK_RXCFG, rxcfg | (0x1 << 28))
e_load("/usr/epiphany/bin/e_mailbox.srec", &dev, 0, 0, E_TRUE);
** interrupt occurs and in the handler - cfg = reg_read(&mailbox, ERX_CFG); 	reg_write(&mailbox, ERX_CFG, cfg & ~MAILBOX_ENABLE);
e_load("/usr/epiphany/bin/e_mailbox.srec", &dev, 0, 0, E_TRUE);
ee_write_esys(ELINK_RXCFG, rxcfg | (0x1 << 28))
e_load("/usr/epiphany/bin/e_mailbox.srec", &dev, 0, 0, E_TRUE);
** interrupt occurs and in the handler - cfg = reg_read(&mailbox, ERX_CFG); 	reg_write(&mailbox, ERX_CFG, cfg & ~MAILBOX_ENABLE);
e_load("/usr/epiphany/bin/e_mailbox.srec", &dev, 0, 0, E_TRUE);
....
int items = ee_read_esys(ELINK_MAILBOXSTAT);
ee_read_esys(ELINK_MAILBOXLO);
```

aolofsson, 2015-12-09:
Just to make sure..the MAILBOXSTAT currently only returns not_empty. I see that you read it into "items". The indicator is binary, you will have to poll between every read to see if it's empty. We can change this. Just wanted to make sure there is no spec confusion.

peteasa, 2015-12-09:
That realization led me to comment "By the way the number of messages in the mailbox queue might be of use."  Anyway I have almost got the epoll event thing working now and I dont think that it depends on a change in entry.  What should happen is that epoll_wait returns when it sees that there is a chance to read a new entry... not sure if it works yet but if not there is a way because commands like tail -f /var/log/messages keep printing the new stuff just added at the end of a file.. same mechanism will allow user side code to see when the interrupt handler has updated the new sysfs file /sys/class/epiphany/epiphany/mailbox_state... more once I have got this going.

aolofsson, 2015-12-10:
Yes, the concept of having fifo count as a readable quantity is great. Less interrupts to service from host side means more messages / sec once we have this working.:-)

peteasa, 2015-12-12:
My driver is now working well and the interrupt from the kernel is working... but I still cant get the e_load to work after the mailbox register has been read.  For now I have commented out the code in the interrupt handler that reads the mailbox.. that way, provided you dont read the mailbox on the user side at least you get the interrupt.  The mailbox host program demos the most odd case:
```
e_load,
enable interrupt,
interrupt
disable interrupt,
read lo / hi
e_load
enable interrupt,
interrupt
disable interrupt,
read lo / hi
e_load .... at this point e_load does not seem to work!!
enable interrupt...
timeout.
```

```
# /usr/epiphany/bin/mailbox
InitialEpiphany(): rows,cols: ( 4, 4)
main(): devfd 6, epollfd 5, kernelEventfd 7, cancelfd 8
RunTest():   0: Message from eCore 0x808 ( 0, 0): RunTest(): pre_stat: 1
RunTest(): mbox_lo: 808
RunTest(): mbox_hi: 810f0320
RunTest(): post_stat: 0
"Hello World from core 0x808!"
RunTest():   1: Message from eCore 0x809 ( 0, 1): RunTest(): pre_stat: 1
RunTest(): mbox_lo: 809
RunTest(): mbox_hi: 810f0320
RunTest(): post_stat: 0
"Hello World from core 0x809!"
RunTest():   2: Message from eCore 0x80a ( 0, 2): RunTest(): pre_stat: 0
RunTest(): mbox_lo: 809
RunTest(): mbox_hi: 810f0320
RunTest(): post_stat: 0

INFO: SleepThenCancel(): Cancelling!
```

peteasa, 2015-12-12:
And I have just found a work round!... Not very pleasant but it works
```
e_reset_system()
e_load,
enable interrupt,
interrupt
disable interrupt,
read lo / hi
e_reset_system()
e_load
enable interrupt,
interrupt
disable interrupt,
read lo / hi
e_reset_system()
e_load .. it continues to work
enable interrupt...
timeout.
```

Not good to have to reset the system but at least it works!

peteasa, 2015-12-12:
And now that I am looking at the hdl I find a potential issue....
```
# /usr/epiphany/bin/mailbox
InitialEpiphany(): rows,cols: ( 4, 4)
main(): devfd 6, epollfd 5, kernelEventfd 7, cancelfd 8
RunTest():   0: Message from eCore 0x808 ( 0, 0): "Hello World from core 0x808!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000808 HI=810f0320
RunTest(): etx_status: f8, erx_status: 808
RunTest():   1: Message from eCore 0x809 ( 0, 1): "Hello World from core 0x809!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000809 HI=810f0320
RunTest(): etx_status: f8, erx_status: 809
RunTest():   2: Message from eCore 0x80a ( 0, 2):
INFO: SleepThenCancel(): Cancelling!
INFO: WaitForMailboxNotifier(): Cancelled!
""
RunTest(): etx_status: f8, erx_status: 809
```

In my failing case I find that erx_status is 0x809 ie rxwr_wait is active.. not that can not be good!  This seems to be why I have to reset the system to get things going again.

peteasa, 2015-12-12:
Now instead of resetting the system I set the first value I read
```
ee_write_esys(ELINK_RXCFG, 0x3e0fe04))
```
and it all works!

So this does seem to be a problem where the ERX_CFG register gets corrupted.. 

```
# /usr/epiphany/bin/mailbox
InitialEpiphany(): rows,cols: ( 4, 4)
main(): devfd 6, epollfd 5, kernelEventfd 7, cancelfd 8
RunTest(): rxcfg register 0x3e0fe04 0x3e0fe04 0x3e0fe04
RunTest():   0: Message from eCore 0x808 ( 0, 0): "Hello World from core 0x808!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000808 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x808, erx_config: 0x3e0fe0c
RunTest(): rxcfg register 0x3e0fe0c 0x3e0fe0c 0x3e0fe0c
RunTest():   1: Message from eCore 0x809 ( 0, 1): "Hello World from core 0x809!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000809 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x809, erx_config: 0x3e0fe0d
RunTest(): rxcfg register 0x3e0fe0d 0x3e0fe0d 0x3e0fe0d
RunTest():   2: Message from eCore 0x80a ( 0, 2): "Hello World from core 0x80a!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=0000080a HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x80a, erx_config: 0x3e0fe0f
RunTest(): rxcfg register 0x3e0fe0f 0x3e0fe0e 0x3e0fe0e
RunTest():   3: Message from eCore 0x80b ( 0, 3): "Hello World from core 0x80b!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=0000080b HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x80b, erx_config: 0x3e0fe0f
RunTest(): rxcfg register 0x3e0fe0f 0x3e0fe0f 0x3e0fe0f
RunTest():   4: Message from eCore 0x848 ( 1, 0): "Hello World from core 0x848!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000848 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x848, erx_config: 0x3e0fe4f
RunTest(): rxcfg register 0x3e0fe4f 0x3e0fe4c 0x3e0fe4c
RunTest():   5: Message from eCore 0x849 ( 1, 1): "Hello World from core 0x849!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000849 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x849, erx_config: 0x3e0fe4d
RunTest(): rxcfg register 0x3e0fe4d 0x3e0fe4d 0x3e0fe4d
RunTest():   6: Message from eCore 0x84a ( 1, 2): "Hello World from core 0x84a!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=0000084a HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x84a, erx_config: 0x3e0fe4f
RunTest(): rxcfg register 0x3e0fe4f 0x3e0fe4e 0x3e0fe4e
RunTest():   7: Message from eCore 0x84b ( 1, 3): "Hello World from core 0x84b!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=0000084b HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x84b, erx_config: 0x3e0fe4f
RunTest(): rxcfg register 0x3e0fe4f 0x3e0fe4f 0x3e0fe4f
RunTest():   8: Message from eCore 0x888 ( 2, 0): "Hello World from core 0x888!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000888 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x888, erx_config: 0x3e0fecf
RunTest(): rxcfg register 0x3e0fecf 0x3e0fe8c 0x3e0fe8c
RunTest():   9: Message from eCore 0x889 ( 2, 1): "Hello World from core 0x889!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=00000889 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x889, erx_config: 0x3e0fe8d
RunTest(): rxcfg register 0x3e0fe8d 0x3e0fe8d 0x3e0fe8d
RunTest():  10: Message from eCore 0x88a ( 2, 2): "Hello World from core 0x88a!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=0000088a HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x88a, erx_config: 0x3e0fe8f
RunTest(): rxcfg register 0x3e0fe8f 0x3e0fe8e 0x3e0fe8e
RunTest():  11: Message from eCore 0x88b ( 2, 3): "Hello World from core 0x88b!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=0000088b HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x88b, erx_config: 0x3e0fe8f
RunTest(): rxcfg register 0x3e0fe8f 0x3e0fe8f 0x3e0fe8f
RunTest():  12: Message from eCore 0x8c8 ( 3, 0): "Hello World from core 0x8c8!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=000008c8 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x8c8, erx_config: 0x3e0fecf
RunTest(): rxcfg register 0x3e0fecf 0x3e0fecc 0x3e0fecc
RunTest():  13: Message from eCore 0x8c9 ( 3, 1): "Hello World from core 0x8c9!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=000008c9 HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x8c9, erx_config: 0x3e0fecd
RunTest(): rxcfg register 0x3e0fecd 0x3e0fecd 0x3e0fecd
RunTest():  14: Message from eCore 0x8ca ( 3, 2): "Hello World from core 0x8ca!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=000008ca HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x8ca, erx_config: 0x3e0fecf
RunTest(): rxcfg register 0x3e0fecf 0x3e0fece 0x3e0fece
RunTest():  15: Message from eCore 0x8cb ( 3, 3): "Hello World from core 0x8cb!"
RunTest(): PRE_STAT=00000001 POST_STAT=00000000 LO=000008cb HI=810f0320
RunTest(): etx_status: 0xf8, etx_config: 0x1, erx_status: 0x8cb, erx_config: 0x3e0fecf
RunTest(): rxcfg register 0x3e0fecf 0x3e0fecf 0x3e0fecf
main(): wait for SleepThenCancel thread to terminate
```

peteasa, 2015-12-19:
Ok so I have created a pull request for you to consider.  I have simulated this and will now try it out on hardware, but the logic is sound.  You may have an alternative way of providing a fix so I have created the pull request now for you to have a look at the proposed fix.

Notice that the OR of 000008cb | 0x3e0fe04 == 0x3e0fecf.. ie both config and emailbox data are active in the ecfg_if.v mi_dout_mux...  My fix filters out the mailbox data thus in the simulation at least fixing the apparent corruption in the register read.

The kernel driver that I have written uses the config register read to do a read modify write to enable / disable the interrupts.  Thus with this bug my kernel driver reads the wrong value from the config register and then writes this whenever the enable / disable of the interrupt is attempted.  This then causes the whole interface to stop functioning... oops!

peteasa, 2015-12-19:
This fix works with hardware also...  I have now found the next issue to investigate.. The new problem occurs without interrupts.  I made a slight modification to your mailbox test where I loop round multiple times rather than just load one core.  In this problem the e-loader fails during e_start(dev, irow, icol).  The board just locks up and no access is possible.. more later when I have had time to investigate this different problem.

aolofsson, 2015-12-19:
Thats great! I will review pr tonight. In regards to lockup, just want to
make sure we are in sync on reser sequence. In order to get thw rx readback
to work reliably, you have yo set the idelay to 7. I dont think this is is
in the regular elib yet. I will send an example.

peteasa, 2015-12-20:
Added a pull request with my changes to the elink/sw/mailbox test.  Also added the driver that I am using today so that anyone can try this out.  The e-main.c file will compile and run with or without the interrupt enabled kernel changes.  The new failure occurs with or without the new interrupt enabled kernel.  To enable interrupts replace the kernel epiphany.c / epiphany.h and rebuild, make sure epiphany.h is in the include path for the e-main.c build and build but change UseInterrupts = 1 in e-main.c and the test will not use a usleep delay but will use interrupts. .. hmm also the device tree needs update adding
<pre>
		elink2_esaxi_0: esaxi@80000000 {
			compatible = "xlnx,parallella-base-1.0";
			interrupt-parent = <&ps7_scugic_0>;
			interrupts = <0 57 4>;
			reg = <0x80000000 0x40000000>;
			xlnx,read-tag-addr = <0x810>;
</pre>

peteasa, 2015-12-21:
closing this issue because it is fixed and the new problem is unrelated to the fix or this issue.

closed 2015-12-21 by peteasa