aolofsson/oh #35: ELINK_MAILBOXSTAT always zero

olajep, 2016-01-18:
not empty flag is never updated

aolofsson, 2016-01-18:
How do you know if the register readback is broken/times out?

olajep, 2016-01-18:
```
$ dmesg
...
Unhandled fault: external abort on non-linefetch (0x1018) at 0xb6984210
```

olajep, 2016-01-18:
... and the program gets aborted with a  bus error signal

olajep, 2016-01-18:
Might be a hint:

https://github.com/parallella/oh/blob/master/emailbox/hdl/emailbox.v#L125
```verilog
//TODO: check that high bits stay high
```

aolofsson, 2016-01-18:
> Might be a hint:
>
> https://github.com/parallella/oh/blob/master/emailbox/hdl/emailbox.v#L125
>
> //TODO: check that high bits stay high

My point was that if the readback is broken, then everything is broken
(including readback from mailbox).

olajep, 2016-01-18:
I can do a 64-bit write to E_MAILBOXLO from Epiphany and then read out the value from ARM via E_MAILBOXLO / E_MAILBOXHI.

aolofsson, 2016-01-18:
That seems very unlikely based on current bug...but I will keep checking.

aolofsson, 2016-01-24:
fixed

closed 2016-01-24 by aolofsson