aolofsson/oh #38: mailbox: Drop writes when FIFO is full

olajep, 2016-01-20:
(read on github)

Writes to mailbox when mailbox is full is destructive. Internal write pointer overflows?
I suggest we just drop writes to the mailbox when `mailbox_full` is asserted (that signal seems to work). Add a sticky (write to clear) overflow bit to `E_MAILBOXSTATUS` to detect. Might also need a `mailbox_half_full` interrupt. Driver can maintain a larger FIFO in RAM (if we need it).

FIFO depth = 32

Ignore 'hi'

32 x `str` = 32 entries can be read back from mailbox until `status.not_empty == 0`
As expected.
```
status: 0x00000003 lo: 00000000 hi: 0x00000660
status: 0x00000001 lo: 0x00000001 hi: 0x810f0730
status: 0x00000001 lo: 0x00000002 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000003 hi: 0x00000003
...
status: 0x00000001 lo: 0x0000001e hi: 0x5555001e
status: 0x00000001 lo: 0x0000001f hi: 0x810f0730
```

33 x `str` = 33 entries read back from mailbox until `status.not_empty == 0`
As can be seen first entry (`lo: 0x00000000`) is overwritten by the last write. Also mailbox_full bit in status is set **after** first readout.
```
status: 0x00000001 lo: 0x00000020 hi: 0x810f0730
status: 0x00000003 lo: 0x00000001 hi: 0xffffffe0
status: 0x00000001 lo: 0x00000002 hi: 0x00000002
status: 0x00000001 lo: 0x00000003 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000004 hi: 0x00000019
status: 0x00000001 lo: 0x00000005 hi: 0x00000005
status: 0x00000001 lo: 0x00000006 hi: 0x00000007
status: 0x00000001 lo: 0x00000007 hi: 0x00000007
status: 0x00000001 lo: 0x00000008 hi: 0x810f0730
status: 0x00000001 lo: 0x00000009 hi: 0xffffffe8
status: 0x00000001 lo: 0x0000000a hi: 0x0000000a
status: 0x00000001 lo: 0x0000000b hi: 0x00007ff0
status: 0x00000001 lo: 0x0000000c hi: 0x00000005
status: 0x00000001 lo: 0x0000000d hi: 0x0000000d
status: 0x00000001 lo: 0x0000000e hi: 0x0000000f
status: 0x00000001 lo: 0x0000000f hi: 0x0000000f
status: 0x00000001 lo: 0x00000010 hi: 0x810f0730
status: 0x00000001 lo: 0x00000011 hi: 0xfffffff0
status: 0x00000001 lo: 0x00000012 hi: 0x00000012
status: 0x00000001 lo: 0x00000013 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000014 hi: 0x0000000d
status: 0x00000001 lo: 0x00000015 hi: 0x00000015
status: 0x00000001 lo: 0x00000016 hi: 0x00000017
status: 0x00000001 lo: 0x00000017 hi: 0x00000017
status: 0x00000001 lo: 0x00000018 hi: 0x810f0730
status: 0x00000001 lo: 0x00000019 hi: 0xfffffff8
status: 0x00000001 lo: 0x0000001a hi: 0x0000001a
status: 0x00000001 lo: 0x0000001b hi: 0x00007ff0
status: 0x00000001 lo: 0x0000001c hi: 0x00000015
status: 0x00000001 lo: 0x0000001d hi: 0x0000001d
status: 0x00000001 lo: 0x0000001e hi: 0x0000001f
status: 0x00000001 lo: 0x0000001f hi: 0x0000001f
status: 0x00000001 lo: 0x00000020 hi: 0x810f0730
```

63 x `str` = 63 entries read back from mailbox until `status.not_empty == 0`
First 31 entries (0x0-0x1e) is overwritten by the last 31 writes (0x20-0x3e)
```
status: 0x00000001 lo: 0x00000020 hi: 0x00000020
status: 0x00000001 lo: 0x00000021 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000022 hi: 0x00000022
status: 0x00000001 lo: 0x00000023 hi: 0x0000001a
status: 0x00000001 lo: 0x00000024 hi: 0x00000024
status: 0x00000001 lo: 0x00000025 hi: 0x0000001c
status: 0x00000001 lo: 0x00000026 hi: 0x00000026
status: 0x00000001 lo: 0x00000027 hi: 0x0000001e
status: 0x00000001 lo: 0x00000028 hi: 0x00000028
status: 0x00000001 lo: 0x00000029 hi: 0x0000002a
status: 0x00000001 lo: 0x0000002a hi: 0x0000002a
status: 0x00000001 lo: 0x0000002b hi: 0x00007ff0
status: 0x00000001 lo: 0x0000002c hi: 0x0000002c
status: 0x00000001 lo: 0x0000002d hi: 0x00000024
status: 0x00000001 lo: 0x0000002e hi: 0x0000002e
status: 0x00000001 lo: 0x0000002f hi: 0x00000026
status: 0x00000001 lo: 0x00000030 hi: 0x00000030
status: 0x00000001 lo: 0x00000031 hi: 0x00000028
status: 0x00000001 lo: 0x00000032 hi: 0x00000032
status: 0x00000001 lo: 0x00000033 hi: 0x00000034
status: 0x00000001 lo: 0x00000034 hi: 0x00000034
status: 0x00000001 lo: 0x00000035 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000036 hi: 0x00000036
status: 0x00000001 lo: 0x00000037 hi: 0x0000002e
status: 0x00000001 lo: 0x00000038 hi: 0x00000038
status: 0x00000001 lo: 0x00000039 hi: 0x00000030
status: 0x00000001 lo: 0x0000003a hi: 0x0000003a
status: 0x00000001 lo: 0x0000003b hi: 0x00000032
status: 0x00000001 lo: 0x0000003c hi: 0x0000003c
status: 0x00000001 lo: 0x0000003d hi: 0x0000003e
status: 0x00000001 lo: 0x0000003e hi: 0x0000003e
status: 0x00000003 lo: 0x0000001f hi: 0x00000020
status: 0x00000001 lo: 0x00000020 hi: 0x00000020
status: 0x00000001 lo: 0x00000021 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000022 hi: 0x00000022
status: 0x00000001 lo: 0x00000023 hi: 0x0000001a
status: 0x00000001 lo: 0x00000024 hi: 0x00000024
status: 0x00000001 lo: 0x00000025 hi: 0x0000001c
status: 0x00000001 lo: 0x00000026 hi: 0x00000026
status: 0x00000001 lo: 0x00000027 hi: 0x0000001e
status: 0x00000001 lo: 0x00000028 hi: 0x00000028
status: 0x00000001 lo: 0x00000029 hi: 0x0000002a
status: 0x00000001 lo: 0x0000002a hi: 0x0000002a
status: 0x00000001 lo: 0x0000002b hi: 0x00007ff0
status: 0x00000001 lo: 0x0000002c hi: 0x0000002c
status: 0x00000001 lo: 0x0000002d hi: 0x00000024
status: 0x00000001 lo: 0x0000002e hi: 0x0000002e
status: 0x00000001 lo: 0x0000002f hi: 0x00000026
status: 0x00000001 lo: 0x00000030 hi: 0x00000030
status: 0x00000001 lo: 0x00000031 hi: 0x00000028
status: 0x00000001 lo: 0x00000032 hi: 0x00000032
status: 0x00000001 lo: 0x00000033 hi: 0x00000034
status: 0x00000001 lo: 0x00000034 hi: 0x00000034
status: 0x00000001 lo: 0x00000035 hi: 0x00007ff0
status: 0x00000001 lo: 0x00000036 hi: 0x00000036
status: 0x00000001 lo: 0x00000037 hi: 0x0000002e
status: 0x00000001 lo: 0x00000038 hi: 0x00000038
status: 0x00000001 lo: 0x00000039 hi: 0x00000030
status: 0x00000001 lo: 0x0000003a hi: 0x0000003a
status: 0x00000001 lo: 0x0000003b hi: 0x00000032
status: 0x00000001 lo: 0x0000003c hi: 0x0000003c
status: 0x00000001 lo: 0x0000003d hi: 0x0000003e
status: 0x00000001 lo: 0x0000003e hi: 0x0000003e
```

64 x `str` = 0 entries read back from mailbox until `status.not_empty == 0`
96 x `str` = 32 entries (0x40-0x5f) read back from mailbox until `status.not_empty == 0`
...

aolofsson, 2016-01-20:
Yes, will add a safeguard on full and agree in regards to making prog_full another possible interrupt_source, but this still needs to be solved in HW by some agreement/API at the end of the day. 
(unless I am missing something). 

Initial thought was to have one entry per core and then have dual sided communication whereby the host would send an ACK (through e_write) back to each core it received a message from. The core ID would be embedded in the message received by the host. The core would not be allowed to send another message until it has received an ACK from host that message has received.

So yes, having fifo full stopping writes is nice, but it doesn't solve the problem.

If we add pushback on fifo full, then yes we could keep pushing data, but if not properly used, this would flood the network quickly....will implement that now.

aolofsson, 2016-01-24:
implemented

closed 2016-01-24 by aolofsson