aolofsson/oh #45: fpga for 7010 device may need reworking

peteasa, 2016-01-30:
ERROR: [Place 30-640] Place Check : This design requires more MMCME2_ADV cells than are available in the target device. This design requires 3 of such cell types but only 2 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.

The above error was seen when attempting to build the hdmi version of the fpga for the xc7z010clg400-1 part.  Note I have added in the ADI hdmi modules so this error is unlikely to be seen when building a headless version.

peteasa, 2016-01-31:
Just to let you know I have just completed a quick test, using hdmi to display a good image on the screen, running the speaker-test to listen to a rather high pitched left / right speaker test PLUS running my interrupt driven emailbox / memory test...

and it all works (kind of).. the corruption on the sound is no doubt because I am using the wrong reference clock speed.  I will experiment a bit but it looks good on a xc7z020clg400-1 part and I have every hope that the xc7z010clg400-1 will also work.

The following output shows the interrupts  f8003000.ps7-dma is I guess sound, 80000000.esaxi is the elink mailbox, hdmi interrupt although wired in is not in the device tree and seems not be be used by the ADI drivers at present and I am not sure why the video dma interrupt is not showing  count but then perhaps I have got something else not quite right or perhaps it is only used with video rather than still images.. anyway a bit of a success thanks to your hard work!

<pre>cat /proc/interrupts
           CPU0       CPU1
 16:          0          0       GIC  27  gt
 17:      13249       8061       GIC  29  twd
 18:          0          0       GIC  43  ttc_clockevent
 22:          0          0       GIC  35  f800c000.ps7-ocmc
 25:          0          0       GIC  45  f8003000.ps7-dma
 26:         37          0       GIC  46  f8003000.ps7-dma
 27:          0          0       GIC  47  f8003000.ps7-dma
 28:          0          0       GIC  48  f8003000.ps7-dma
 29:          0          0       GIC  49  f8003000.ps7-dma
 30:          0          0       GIC  72  f8003000.ps7-dma
 31:          0          0       GIC  73  f8003000.ps7-dma
 32:          0          0       GIC  74  f8003000.ps7-dma
 33:          0          0       GIC  75  f8003000.ps7-dma
 34:          0          0       GIC  40  f8007000.ps7-dev-cfg
 35:      29942          0       GIC  54  eth0
160:       7776          0       GIC  79  mmc0
167:        989          0       GIC  82  xuartps
168:        445          0       GIC  57  cdns-i2c
172:        167          0       GIC  53  e0002000.ps7-usb
174:         43          0       GIC  39  f8007100.ps7-xadc
175:          0          0       GIC  91  xilinx-dma-controller
176:        100          0       GIC  89  80000000.esaxi
IPI1:          0          0  Timer broadcast interrupts
IPI2:       3661       9596  Rescheduling interrupts
IPI3:          0          0  Function call interrupts
IPI4:         24         30  Single function call interrupts
IPI5:          0          0  CPU stop interrupts
IPI6:          0          0  IRQ work interrupts
IPI7:          0          0  completion interrupts
Err:          0
</pre>

aolofsson, 2016-02-01:
Very cool! I am looking into clocking to see if we can use 1 less pll.

peteasa, 2016-02-05:
Perhaps scarce resource should be outside the main block so that a user may use other outputs from the same clocking source?

peteasa, 2016-02-07:
Sound on 7010 - Snookered!  I dont want to change the boot loader so I am stuck with the FSBL configured clock speeds.  Never mind.  At least the 7010 board hdmi has a chance of working now that the oh fpga builds for that.