aolofsson/oh #47: RX: Dynamic mapping issues

olajep, 2016-02-02:
With this commit:
https://github.com/parallella/oh/commit/5dc9375ebb5f41a890e10d23dbe25ec4ccc6c85e

### Setup

```
ELINK_RXCFG = { .mmu_enable = 1, .remap_mode = 2} = 0xA
```

Memory mappings written to :ELINK_RXMMU table
```

ERAM [0x8e000000-90000000] --> [0x3e000000-0x40000000]
addr=0x810ec700 val=0x000003e0
addr=0x810ec704 val=0x00000000
addr=0x810ec708 val=0x000003e1
addr=0x810ec70c val=0x00000000
addr=0x810ec710 val=0x000003e2
addr=0x810ec714 val=0x00000000
...
addr=0x810ec7f0 val=0x000003fe
addr=0x810ec7f4 val=0x00000000
addr=0x810ec7f8 val=0x000003ff
addr=0x810ec7fc val=0x00000000

FPGA ELINK (so Epiphany can access mailbox regs)
addr=0x810ec080 val=0x00000810
addr=0x810ec084 val=0x00000000
```

### What happens

Writes from Epiphany to ERAM does not work.
Reads from Epiphany to ERAM returns something (not random, different value from different addresses), but the wrong data.
Writes to ELINK_MAILBOXLO does work, **iff** the FPGA elink is present in the RX MMU table. In fact if I change the FPGA elink entry in the MMU table to an ERAM address then writes from Epiphany to ELINK_MAILBOXLO.is written to ERAM
NB: Entries in ELINK_RXMMU not cleared on reset via ELINK_RESET (which is fine, can do that in S/W)

olajep, 2016-02-03:
Ok found it, guess I should read the RTFM first!
https://github.com/parallella/oh/blob/master/elink/hdl/erx_remap.v#L55-L58
```verilog
   //more complex compresssed map
   assign dynamic_remap[31:0] = addr_in[31:0]    //input
			     - (colid << 20)     //subtracing elink (start at 0)
			     + remap_base[31:0]  //adding back base
                             - (addr_in[31:26]<<$clog2(colid));
```
... and then this is fed into erx_mmu.

But please explain what's the idea here, don't get it?

aolofsson, 2016-02-03:
What does this have to do with the MMU?

olajep, 2016-02-03:
emesh_remap_packet is input to erx_mmu
https://github.com/parallella/oh/blob/master/elink/hdl/erx_core.v#L163

olajep, 2016-02-03:
Wrong:
ELINK_RXCFG = { .mmu_enable = 1, .remap_mode = 2} = 0xA

Right
ELINK_RXCFG = { .mmu_enable = 1, .remap_mode = 0} = 0X2

Ok, got it

olajep, 2016-02-04:
Seems to work :) . Switched to mmu mapping on my driver devel branch.

closed 2016-02-04 by olajep