aolofsson/oh #98: system_bd.tcl Vivado version upgrade gotcha

olajep, 2016-11-28:
In fpga/parallella/headless_e16_z70[12]0/system_bd.tcl:

When upgrading Vivado, you must change the line with:

```tcl
set scripts_vivado_version 2015.4
```

to match the Vivado version. Even though this file is re-generated in the
build, it is required by the build to create an initial block design.

Ugh!!!

promach, 2017-06-09:
I am using Vivado version 2017.1   

I have followed the instructions in [https://github.com/parallella/oh/issues/103#issuecomment-298262178](https://github.com/parallella/oh/issues/103#issuecomment-298262178) but I still have the following error. 

> [ERROR]  : Can't read BIT file - ./system.runs/impl_1/system_wrapper.bit 
cp: cannot stat 'system_wrapper.bit.bin': No such file or directory

I have attached the error log file below.

[error.txt](https://github.com/parallella/oh/files/1063479/error.txt)

promach, 2017-06-12:
When I open the system.xpr in GUI, I have this warning in synthesis:

> [Project 1-486] Could not resolve non-primitive black box cell 'system' instantiated as 'system_i' ["/home/promach/Documents/Grive/Personal/Digital/Parallella/oh/src/accelerator/fpga/system.srcs/sources_1/bd/system/hdl/system_wrapper.v":16]