aolofsson/oh #117: packet2emesh can not be find

liyd20220105, 2024-12-12:
Hi Andreas Olofsson,

I am a beginer for this area. I find many blocks use packet2emesh. But I can not find it. could you please upload it or tell me how to find it. Many thanks.

peteasa, 2025-06-28:
look in the git history!  as a beginner I would not start with the latest and greatest.  start with a stable version.  today I am working with https://github.com/peteasa/oh.git commit cc24256 that is based on Ola's fork of this repo.  i also have a version on gitlab that I am using for [PYNQ-e](https://paracpg.gitlab.io/) but that is not as stable as Ola's fork.

liyd20220105, 2025-06-28:
Thank you! After I posted the comment, I searched and found this module. I really appreciate you providing the link to the stable version.

peteasa, 2025-06-28:
no problem, let me know how you get on!